LilyGO/TTGO-T8-ESP32 #1: On-board LEDs

elf55, 2018-04-29:
There are at least 3 on-board LEDs.  Is there documentation on how they are connected to GPIO pins?

LilyGO, 2018-05-24:
One is a power light, and the other is an LED foot controlled by the GPIO21 pin.

LilyGO, 2018-05-24:
If you need it, we can send you a T8 schematic

amirgon, 2018-05-24:
What would happen to the power light during deep-sleep? Would it still be on?

elf55, 2018-05-24:
A schematic would solve all questions, but knowing what the LEDs are connected to helps immediately.

closed 2018-05-24 by elf55

pilnikov, 2018-08-14:
I need a schematic too
gpio 16, 17 where?

remedix, 2021-01-20:
Can you please explain how to turn off these LEDs ? There is one blinking LED and one constant on LED

pilnikov, 2021-01-21:
just unsolder

johndoesenior, 2021-02-04:
I need the schematic also.
I cant find TX2,RX2 in the pinout