LilyGO/TTGO-T8-ESP32 #3: red/blue LEDs

carstenblt, 2018-08-07:
Hi there,
there are two LEDs on the board that I find are not connected to any GPIO. Am I just missing the right one? Any way to disable them via software?

Thanks in advance.

wieb18, 2019-02-03:
There is only 2 LEDs on this board, BLUE (power indicator) and GREEN (wifi indicator).
The GREEN LED can be controlled with GPIO21.
To turn off the BLUE LED, power the board through Lithium battery interface.

heeen, 2019-05-09:
the blue LED is connected to the battery charging controller, it goes off if the battery is not charging
the red LED is supposed to light when the board is powered, but it is not working for me, neither from USB, nor from a battery

heeen, 2019-05-09:
@LilyGO is it a known issue that the red led is not working?

PureBasic, 2019-05-10:
Wait, there is a red led ?
Doesn't work on my board neither then.

PureBasic, 2019-05-10:
Okay, just looked at it closely, the red led (LED_G2) is implanted but R22 (the 1k brake) is not. 
If you want the red led to work you will need to solder the 0402 resistor. I believe they didn't implanted it to avoid draining batteries continuously. Better to blink LED_G1 (IO21) when needed

stanleyseow, 2020-05-19:
Where did you find the schematics for TTGO T18 V3.0 ???

PureBasic, 2020-05-19:
What is the T18?

patrickelectric, 2023-04-20:
My RED led is blinking like crazy, how do I control it ?