LilyGO/TTGO-T8-ESP32 #5: Schematic

rochba, 2019-02-05:
Hello,
I'm working with these cards and I have a lot of questions about battery management, leds connectivity, antenna …
Is it possible to have access to the schematic. That will solve my issue.
Thanks,
Best regards,

ThomasChr, 2019-05-07:
This is the shematic: https://github.com/LilyGO/TTGO-T8-ESP32/blob/master/Schematic.pdf

Maslenikoff73, 2021-01-30:
> This is the shematic: https://github.com/LilyGO/TTGO-T8-ESP32/blob/master/Schematic.pdf
link not work :(

ThomasChr, 2021-01-30:
Try this one: https://github.com/LilyGO/TTGO-T8-ESP32/raw/master/t8_v1.7.1.pdf

Cyb3r-N1nja, 2022-02-28:
Does anyone here have a working source code to display something to the Display because in my case it says written successfully and the screen remains black