LilyGO/TTGO-T8-ESP32 #6: Which board should i use?

sync5, 2019-02-06:
I'm not sure which esp32 board from the Arduino IDE should I use for this ttgo-t8 1.3? Thanks.

knopserl, 2019-03-17:
same question here: TTGO T8 v1.7. it would be nice and required if a board definition is provided for Arduino Studio.
We always have o use another board like WEMOS LOLIN32 and do a lot of manual works to redefine and add the additional peripherals like the SD Cards etc.

m-menon, 2019-04-11:
> I'm not sure which esp32 board from the Arduino IDE should I use for this ttgo-t8 1.3? Thanks.

LOLIN D32 PRO seems to work for me. I had my board running only for a short while so I haven't tested it completely. The only thing I tested was a I2C OLED on pins 21/22 (SDA/SCL).