LilyGO/TTGO-T8-ESP32 #7: microSD data pins

kaofishy, 2019-03-21:
The pinout for all versions of the board states that the microSD data pins are correctly connected to the SDMMC host interface on the ESP32, but in the schematics it shows DAT1-3 as disconnected and only DAT0 connected. On the board I have I can only make SDMMC work in 1-bit mode so it suggests to me the schematics is correct. Please confirm whether this is the case or not. Thanks.

tionebrr, 2020-07-22:
The sd slot is wired in SPI 1-bit mode.

venice1200, 2021-03-20:
I can confirm working 1-Bit SD-Bus-Mode and SPI Mode.
But I have to **SD_MMC.begin("/sdcard", true)** twice in my Arduino Sketch.
First **.begin** fails, second is OK.

koradeh, 2024-06-01:
`// pullup GPIO2 for SD_MMC mode, you need 1-15kOm resistor connected to GPIO2 and GPIO19 `
Or
`pinMode(2, INPUT_PULLUP);`