LilyGO/TTGO-T8-ESP32 #8: Deepsleep only 3.5 mA

ThomasChr, 2019-05-15:
When I put the ESP32 in deepsleep I still have a power draw of ~3.5mA.
Can you explain why is that so, and can I do anything about it?

ThomasChr, 2019-05-17:
I‘ve desoldered the SIL2104 and now the power consumption is down to 3.1 mA, better, but far from the uA the ESP32 is capable of!

ThomasChr, 2019-05-20:
The problems seems to be that VDD_SDIO stays at 3.3V in deepsleep, so flash and psram do not get powered down.

ThomasChr, 2019-05-20:
Some news here: 
1. VDD_SDIO does not go low when entering deepsleep. It seems that the Resistor R20 is placed on the board and pulling VDD_SDIO to 3.3V. Removing it helped and I lost one more mA (1.5 mA total in deepsleep)
2. I powered the board directly on the 3.3V rail, so the SYS8008B Voltage regulator was not involved and I'm down to 295 µA.

So still there are some 280 µA to catch. Anyone got an idea what I can try now?
Of course, everything is allowed, even experiments which will destroy the board, all to satisfy my curiosity!

ThomasChr, 2019-05-21:
After blowing the power IC completely from the board the power consumption is down to 100 µA - getting there! Any hints what should be blown off next? Some pullup resistors maybe?

PureBasic, 2019-05-21:
Wow, that was a very leaky regulator...
You could just check the voltage across all the remaining resistors. R1, R4, R11, R12, R13, R14 and R19. Also there is 1R3 and 1R12 voltage divider close to the SYS8008B if still connected draining a bit for sure.
It may also be a leaky cap somewhere, which is harder to find.

ThomasChr, 2019-05-21:
Hi PureBasic, thank you very much for your answer!

Is there anywhere a Layout of the board? The resistors are not marked on the PCB which makes it very hard to find the right one!

PureBasic, 2019-05-22:
Arf you will have to follow the schematic, count the pins and so on... There is no TOPCOMP sheet from what I gathered.
1R3 and 1R12 should be close the SYS8008, they are draining 6uA (553k @ 3.3v).

How much exceding current do you still have to find ? You are feeding power directly to the 3v3 net or VBat ?

ThomasChr, 2019-05-22:
Hello PureBasic, 

I just blowed some resistors of the board and this is the outcome:
1. Without voltage regulator (providing 3.3V directly on the rail): 100 µA
2. Without the Voltage Divider from the Power IC: 90 µA
3. Without the pullups for the SD Card IO Lines: 5 µA

So I think I'm pretty much done :-)

PureBasic, 2019-05-22:
Nice ! 
It's weird that the SD pullups are draining that much current, because they are marked as NC on the schematic, so maybe you actually removed the ESD protections 😆.
Did you checked the state of the IOs of the ESP32 when it goes in hibernation ? They should all be put in "INPUT" mode with no internal pullups or pulldowns, so virtually connected to nothing inside the ESP32.

Civilduino, 2019-07-22:
And NC means Not Connected?

PureBasic, 2019-07-22:
Yes. But check on the board, you never know...

hpsaturn, 2020-04-14:
Nice thread, guys maybe you know what is the better TTGO board "out the box" with this improvements? thats is, what board has the better performance for hibernation without any hack?
For example TTGO-T-Writsband in my tests reached ~3 days only on suspend mode (not bad because the battery is very tiny), maybe the TTGO T-Display will be? (I'm waiting for a one board for starting tests)