LilyGO/TTGO-T8-ESP32 #9: External antenna TTGO T8

Civilduino, 2019-07-22:
I got the TTGO T8 V1.7 20180721 and I will like to use the external antenna connector. Is it just to plug the antenna?

I tried that I don't think it is working. Do I need to solder some resistor, cut a trace? add a jumper?

Without the antenna the board is useless for my application.

Thank you for clarification and improving the documentation. Some pictures of what to do should be good.

KuhlTime, 2020-06-27:
I am having the same question. Looking at the schematic I would assume that you have to desolder R16 and bridge R23 (or use the desoldered 0Ω resistor). It would be nice to get some clarification on this though.

[image]

vaughnlfisher, 2021-03-26:
Did you ever get an answer to this

LiveRock, 2021-07-02:
Any answer to this? I bridge R23 with a wire (not resistor) and the performance because much much worse even with an external antenna...

KuhlTime, 2021-07-04:
Sadly I did not get an answer. My performance decreased as well when I tried that.

KuhlTime, 2021-07-04:
Maybe good to point out that there is another question regarding the IPEX antenna: https://github.com/LilyGO/TTGO-T8-ESP32/issues/20