LilyGO/TTGO-T8-ESP32 #13: Battery measurement

eried, 2020-03-15:
Hi!

I just wanted to add that it would be useful to have the following info in the repo:

1) PIN 35 measures the VBAT (or USB) with a resistor divider bundled in the board.

```
    #define PIN_VBAT 35
    int v = analogRead(PIN_VBAT); // This returns the battery level
```

2) VCC does not provide any power if we use the battery (I do not know why, but it would be useful to change this in the pcb for next revision and provide USB/VBAT)

paolovr70, 2020-03-26:
Eried I have T8 1.7.1 board. Analog read finds a value in PIN35 around 2430. This level is present both with BATTERY connected and also with NO BATTERY. Are you sure that it correspond to V.bat?

eried, 2020-03-26:
Yes, in that case is USB voltage

paolovr70, 2020-03-27:
But what is the scale of 0-4095.measurement? Because lithium battery is 3.0-4.2v. Board and pins volt are maximum 3.3v and 0-4095 usually are for 0-3.3v. How.did you manage that?

eried, 2020-03-27:
Well, I didnt exactly needed the voltage value, just the reference (because the analogRead changes for every input responding to VBAT, so the values are all different if you are on battery power vs USB)

So what I did is to map the range:
https://github.com/eried/pidflight-lap-firmware/blob/master/pidflight_lap/rx5808.ino#L37

For a simple mapping I would use:

    #define PIN_VBAT 35
    double v = analogRead(PIN_VBAT)  / 4096 / 2 / 4.2 ; // This returns the voltage (max range / voltage divider / max battery voltage )
    // voltage divider = in the PCB of the T8 you can see there is 2 equal resistors, hence 50% division
    // max battery voltage = this is the output of the lipo charging circuitry before it gets to the 3.3 regulator... this circuit is before the ON/OFFswitch)

For a more accurate reading, I would calibrate a mapping with few known voltages and an polinomial interpolation in excel.

eried, 2020-03-27:
Also, consider that while reading the serial monitor via USB you will always get the same VBAT. Create a sketch with a simple bluetooth serial and try with different batteries.