LilyGO/TTGO-T8-ESP32 #14: LEDS on TTGO ESP32 T8 1.7.1

paolovr70, 2020-03-26:
Dear All,
where can I find a pinout of the board? (on the site producer there is no schematic).
There are 3 leds on board. One is for battery, but the other two?
Are there problems if I need to use I2C and one of the leds is connected to 21/22 pins (which are the hardware I2C pins)?
Thanks.
paolo

tionebrr, 2020-07-22:
I Paolo. 
- According to the schematic, blue is for battery charge, red is the 3v3 power, and green is wired to IO21.
- Schematic is in this repository, but the version changes are not pushed here. The pinout is the same as other official ESP32 modules.
- You can use any pin as I²C with the ESP32. GPIOs are actually very flexible. Refer to the ESP32 datasheet.

There is not a lot of help or information about this specific module (the T8) because there is nothing special about it really. It is wired like most other ESP32 modules and thus can be used like any other module. Only difference is the SDCard slot which is actually wired in SPI mode like shown on the schematic and not like advertised.

hasaranga, 2021-01-21:
> 
> 
> Hi Paolo.
> 
>     * According to the schematic, blue is for battery charge, red is the 3v3 power, and green is wired to IO21.
> 
>     * Schematic is in this repository, but the version changes are not pushed here. The pinout is the same as other official ESP32 modules.
> 
>     * You can use any pin as I²C with the ESP32. GPIOs are actually very flexible. Refer to the ESP32 datasheet.
> 
> 
> There is not a lot of help or information about this specific module (the T8) because there is nothing special about it really. It is wired like most other ESP32 modules and thus can be used like any other module. Only difference is the SDCard slot which is actually wired in SPI mode like shown on the schematic and not like advertised.

Can it run in 1bit mode? Because I have the same board but the first call to **SD_MMC.begin("/sdcard", true)** always fail. Second call works. Tried with a different SD card and reduced the cpu freq but the same results.

[image]

tionebrr, 2021-01-22:
I'm using the ESP-IDF and thus don't know anything about the specifics of the arduino implementation.
Remember, the SDCard is wired as an SPI device, so it is one bit. There might be a way to tell SD_MMC to initialize as an SPI SD.

dizcza, 2021-09-03:
@tionebrr could you please clarify this

> Remember, the SDCard is wired as an SPI device, so it is one bit

aren't you trying to say that an SD card will work only in 1-bit width mode for this board while using the SD-MMC driver?

tionebrr, 2021-09-03:
@dizcza Good question. 
The SDCard socket is not wired on SDMMC pins. So you can't have a 1-bit SDMMC. 
I used the [SDSPI Host Driver](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sdspi_host.html) to make it work.

tionebrr, 2021-09-03:
@dizcza You're right, sorry for the confusion. I re-checked the docs and it should work in SDMMC 1-bit mode. I edited my previous comment.

dizcza, 2021-09-03:
@tionebrr thank you for the clarification. It did help because I thought I was doing something wrong by not being able to mount an SD card with the 4-bit operation mode.

BTW, shouldn't this issue be closed? It has been answered fairly in the first comment. The author perhaps was wondering whether the LED will be blinking if IO21 pin is used as SDA with some I2C-connected device. In either case, you can't use it as LED with active I2C communication going through this pin.

Let me also answer another question posted here to quantify it's fully resolved.
@hasaranga you had the following problem:
> ... I have the same board but the first call to SD_MMC.begin("/sdcard", true) always fail. Second call works.

It's because a pull-up resistor is missing on D0 line (as in many other variants of an ESP32 board). You can do either a hardware fix (pulling up a 10 kOhm resistor to pin "2") or a software fix. The software fix is to call `pinMode(2, INPUT_PULLUP);` prior to initializing the SD card. Below is a working example of using SDMMC driver with TTGO T8 in Arduino IDE with the latest esp32 library installed (v1.0.6; I've not checked v2.0.0 yet).

<details>
    <summary>An example of using SDMMC driver with TTGO T8 in Arduino IDE</summary>

```c
/*
 * Connect the SD card to the following pins:
 *
 * SD Card | ESP32
 *    D2       12
 *    D3       13
 *    CMD      15
 *    VSS      GND
 *    VDD      3.3V
 *    CLK      14
 *    VSS      GND
 *    D0       2  (add 1K pull up after flashing)
 *    D1       4
 */

#include "FS.h"
#include "SD_MMC.h"

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.name(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void createDir(fs::FS &fs, const char * path){
    Serial.printf("Creating Dir: %s\n", path);
    if(fs.mkdir(path)){
        Serial.println("Dir created");
    } else {
        Serial.println("mkdir failed");
    }
[... 141 lines cut ...]
    uint64_t cardSize = SD_MMC.cardSize() / (1024 * 1024);
    Serial.printf("SD_MMC Card Size: %lluMB\n", cardSize);

    listDir(SD_MMC, "/", 0);
    createDir(SD_MMC, "/mydir");
    listDir(SD_MMC, "/", 0);
    removeDir(SD_MMC, "/mydir");
    listDir(SD_MMC, "/", 2);
    writeFile(SD_MMC, "/hello.txt", "Hello ");
    appendFile(SD_MMC, "/hello.txt", "World!\n");
    readFile(SD_MMC, "/hello.txt");
    deleteFile(SD_MMC, "/foo.txt");
    renameFile(SD_MMC, "/hello.txt", "/foo.txt");
    readFile(SD_MMC, "/foo.txt");
    testFileIO(SD_MMC, "/test.txt");
    Serial.printf("Total space: %lluMB\n", SD_MMC.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD_MMC.usedBytes() / (1024 * 1024));
}

void loop(){

}
```

</details>

dizcza, 2021-09-04:
@tionebrr 
could you also clarify
> You can use any pin as I²C with the ESP32

Clearly, we cannot use *any* GPIO pin as I2C, can we? There are only two I2C according to the seller information. One is at 21 & 22 pins. Where is the other? (I'd like to keep pin 21 for LED.)
Thank you.

tionebrr, 2021-09-04:
@dizcza Not really any indeed. You can't use a "input only" pin for SDA for example. 
There are two I2C peripherals in the ESP32 but they are behind a GPIO Mux. You can route most ESP32 internal peripherals to any pins. The GPIO Mux might introduce timing issues with high speed stuff (because it's clocked by the APB at 80MHz), so some peripherals are better used with their default pins. I2C is obviously tolerant enough.
You can check the list in the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf) page 37.

Also I agree that this issue should be closed. But the LilyGo folks does not seem to be monitoring this repository at all.