LilyGO/TTGO-T8-ESP32 #18: schematic for TTGO-T8-ESP32 v1.8

javarobotic, 2020-09-13:
Hi, 

I got the new esp32 T8 v1.8 below:
https://www.aliexpress.com/item/32879370278.html

Wondering if there is really a TTGO-T8-ESP32 v1.8? I cannot find any information about correct pinout and schematic.
Is there any accurate pinout diagram or schematic available? The pinout mentioned in the store is wrong.

fritzlaszlo, 2022-02-22:
Hi, I am having the same problem. @LilyGO, if you could send me a schematic, that would help a lot. Thank you!

John-Tito, 2022-03-07:
> Hi, I am having the same problem. @LilyGO, if you could send me a schematic, that would help a lot. Thank you!

这是你正在找的东西吗？
is this what u are looking for？
https://github.com/Xinyuan-LilyGO/LILYGO-T8

它就在商品介绍页面中。
It is just in the product introduction page
https://www.aliexpress.com/item/32879370278.html

HankLloydRight, 2022-07-06:
Thank you for the schematic.  
If I am supplying +5v power to the board not via the micro-USB jack, do I send +5v to the VIN(VBUS) pin #16 or the +5V pin #15?