LilyGO/TTGO-T8-ESP32 #20: Hows to use IPEX antenna

pidou46, 2020-10-28:
Please, could you give instructions how to use IPEX antenna instead of 3D antenna built in ?

dgoadby, 2020-10-29:
The circuit diagram shows two 0 ohm resistors (aka links). To use an external antenna via the Apex socket you need to remove R11 and add a link to R10. Note: not all of my LilyGO's have the Apex socket fitted. If the Apex socket is not fitted then the pads are usually there to add one. Here is a useful link for a different board but the idea is the same: [Using the Apex antenna socket](https://randomnerdtutorials.com/esp32-cam-connect-external-antenna/.)

pidou46, 2020-10-30:
Thank you for your answer.

I have two board with different revision.
For one of it it's obvious I see the circuit make a Y and it's easy  two understand wish resistor need to be removed and where to make the link.

But for the second one, there is a bunch of 4 resistors and I'm unable to identify wish one to remove and where to solder!
I try to make a picture but failed to get a nice one.

I suggest for the next release, that you add it on the pin layout to avoid recurrent questions about it.

pidou46, 2020-10-30:
My board is T8 V1.1
Here is what I think I have to do:
[image]
Are you agree ?

visicon-masc, 2021-04-06:
> I will research this some more as I too need to use an external antenna. For now I hope that I have been of some help.

Any news on this topic? Has anyone any expirience in using IPEX antenna without populating C17 and L4?
Has anyone gained information about values for C17 and L4 for correct matching? (Talking about v1.7)

LiveRock, 2021-07-05:
I asked my supplier and he said only a resistor bridge is needed. I tried but it does not work. I am asking him to double check with the manufacturer. Thanks

heol-files, 2021-08-12:
Any news on this?
After looking at my boards the above suggestion that I have to add extra capacitors/Inductors after moving the link R16 -> R23 seems to be the way to go to me.

heol-files, 2021-08-12:
@LiveRock Have not tried it yet, missing components. But I have IPEX connectors so I will try mount one on a TTGO T7_V1.4 instead and move the link R15 -> R14 to see if that works.
Last picture in this thread:
https://github.com/LilyGO/TTGO-T7-Demo/issues/3